prusa3d/Prusa-Firmware-Buddy #4839: [BUG]

wizzlebippi, 2025-09-17:
### Printer model

Core One

### Firmware version

6.40 alpha 1

### Upgrades and modifications

MMU3

### Printing from...

USB

### Describe the bug

MMU toggle state not saved when preserving HW config during full reset. 

### How to reproduce

Conduct a full reset preserving HW config with MMU3 connected.  Post reset, the MMU is disabled in settings.

CZDanol, 2025-09-19:
Hello, whether the MMU is enabled or not is not considered a hardware configuration by us, but a feature of the printer.

wizzlebippi, 2025-09-19:
How is the MMU not hardware? This should be no different than advanced filtration, the buddy camera power, or a hardened nozzle, all of which are kept configured by saving the HW config during reset. Either all these things are features, or they're all hardware. As is, it's confusing and a bug.

closed 2025-11-26 by github-actions[bot]