prusa3d/Prusa-Firmware-Buddy #4840: [BUG]

s3anami, 2025-09-17:
### Printer model

XL/Multitool

### Firmware version

6.4.0-alpha+11774

### Printing from...

Prusa Connect

### Describe the bug

[image]
Printer starts with an unknown error

### How to reproduce

Turn on printer?

### Expected behavior

Printer actually work

3d-gussner, 2025-09-18:
@s3anami Thanks for testing 6.4.0-alpha. Can you please check if you have a dump file you can share. LCD -> Settings -> System -> Save Crash Dump and send it reports@prusa3d.com

After the reboot this never happened again?

s3anami, 2025-09-18:
It hasn't happened again yet. I can try to get a crash dump. I was told to try this by support to help with horrible VFAs I experience with the XL. It hasn't really done anything even when the firmware manages to work which is rarely.

3d-gussner, 2025-09-22:
Can you please connect via USB and use serial terminal like [Putty](https://help.prusa3d.com/article/crash-dump_364959) with 57600 baud rate and connect just after "Bootloader 2.5.0" disappears and LCD output switches to "Firmware ..." and share the log files.

schorsch3000, 2025-09-22:
@3d-gussner i would if i could. but on Friday that bug just dissapeared. And was never seen again, is didn't do anything but printing stuff and turning the printer of and on again.

Or would that be usefull even if it's not in an error-state?

schorsch3000, 2025-09-22:
yes, its gone no for no apparent reasons.

It came up after the first cold boot after flashing and stayed for at least 10 cold boots, every time, but pressing reset brought backt the machine every time.

On 

Since Friday night its just gone, its not flaky, i've tried booting 20+ times this morning from a complete powerloss and it's just gone now.

if the log of my, now working machine, helps, i happily create that for you, but i'm positive, a log from a working boot will not provide evidence for a failure that doesn't trigger anymore.

i'm still on the alpha, and if that error ever comes up again i try to get a log and report it.

As a software developer i know how bad flaky bugs like that can be :-(

Gladly the workaround for the user is as simple as it could be