prusa3d/Prusa-Firmware-Buddy #4842: [BUG] Head crashes, Filament stuck and blue screen of death crashes with 6.4.0 Alpha on XL

Mr3Deee, 2025-09-17:
### Printer model

XL Multi tool

### Firmware version

6.4.0 Alpha

### Upgrades and modifications

Exterior case (EnXlosure)

### Printing from...

PrusaConnect

### Describe the bug

[Started testing for BFW5841]

**TLDR Problems:**
1. Upon pre-purging the toolparking seems to use a differnet algorithm which causes a head crash every time for me. Other then that all other park/pick operations are fine.
2. I see constantly "filament stuck" upon head change to #2. What can I do to mitigate that? 
3. I saw 2 Blue screen of deaths. I can send the files if you want to investigate.

**The details**
I installed the new alpha bios yesterday. Did do the home calibration and tried to print a model with 2 colors, whereas the minor color was in Head #1 (Text) and the main model itself was in Head #2. The model itself takes about 1.5 hours to print. Nothing Special as I have printed this over the past 2 days 2 times already.

1. Calibrated Home.
2. Started to print. Head #1 was pre-purging at the sheet.
3. Going back home to pick up Head #2 produced a head crash as the head was not parked properly (never happened before)!
4. I fixed the crash and it was re-homing. Then picked up Head #2 and started drawing.
5. Then - when head changes happened, I didn't see any new head crashes - all was good.
6. However - every layer change to Head #2 was producing a noise from the head and then a "filament stuck" message. (I guess that the retracting has problems there?) I couldn't observe this on Head #1  - only on #2. But it was consisten and every layer needed a 5 times click to restart printing.

Today I tried it again:

1. Tried to pick and park tools via menu. Did do that at least 30 times and there were no problems. All good.
2. Re-calibrated Home once more as well as other things.
3. Tried again pick and park - still - all good.
4. Started to print the same thing again with two other heads to exclude the possibility of the head being the problem. The result: After the pre-purging of the filament another head crash when switching to the head which does the first printing to do the pre-purge. And then - a Blue screen of death. Never have seen that. I have stored the dump on my USB drive and the printer rebooted. Just to get another blue screen of death. 

I am not sure what to do now. Any help is appreiciated!

### How to reproduce

Please see in description.

Install latest Alpha BIOS and try to print. Does not work for me

### Expected behavior

It prints - just like it always does.

murk-sy, 2025-09-18:
I would recommend downgrading since head crashes seem like a common issue. Haven't seen anything for filament stuck so far, though.

You can do it from the connect interface or following this guide:
https://help.prusa3d.com/article/how-to-downgrade-firmware-core-one-mk4-s-mk3-9-s-mk3-5-s-xl_725930

Mr3Deee, 2025-09-18:
Thaks for your comment! I did downgrade already. I guess that that will take care of the docking problems as well as the BSOD's.

Does anyone need the BSOD's?

However - I do have the feeling that the "Filament stuck" might be something tool specific (even though I haven't seen that with previous BIOS versions). But the fact that I hear some scraping there before the head comes forward shows me that maybe there is something I can do about it. But what?

Any hint? Should I try to fiddle around with the idler screws?

murk-sy, 2025-09-18:
If you mean scraping sounds during tool pickup, then doing dock calibration for that tool or [lubricating the coupler pins](https://help.prusa3d.com/guide/how-to-lubricate-the-coupler-pins-on-original-prusa-xl-multi-tool_636777) may solve that. But if you hear that while extruding/retracting, even when the door is disengaged, it could be an issue with the extruder itself.

I'd say keep the issue open so the "Filament stuck" can be noted by developers.

Mr3Deee, 2025-09-18:
Okay. 

No, the "scraping" came from the (Idler?) when it wanted to pull out the filament but couldn't - probably for auto retract reasons. It couldn't do it and so it told me that the filament was stuck.

I could only see two things which could do this:
a. the new auto retraction (or insertation?) is trying to engage when the material isn't warm enough - or -
b. The idler is not able to move the filament due to a mechanical problem in the ... idler? However - as I didn't observe a problem with inserting or ejecting of filament in earlier BIOS versions (well - maybe once or twice in total?) I'd rather think that this is not it. 

See my point 6 from above:
> However - every layer change to Head 2 was producing a noise from the head and then a "filament stuck" message. 
> (I guess that the retracting has problems there?) I couldn't observe this on Head 1 - only on 2. 
> But it was consisten and every layer needed a 5 times click to restart printing.

Well - if anyone is interested I can give a more detailed description.

MaxGit001, 2025-09-21:
> ### Printer model
> XL Multi tool
> 
> ### Firmware version
> 6.4.0 Alpha
> 
> ### Upgrades and modifications
> Exterior case (EnXlosure)
> 
> ### Printing from...
> PrusaConnect
> 
> ### Describe the bug
> [Started testing for BFW5841]
> 
> **TLDR Problems:**
> 
> 1. Upon pre-purging the toolparking seems to use a differnet algorithm which causes a head crash every time for me. Other then that all other park/pick operations are fine.
> 2. I see constantly "filament stuck" upon head change to [Fix line endings. #2](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2). What can I do to mitigate that?
> 3. I saw 2 Blue screen of deaths. I can send the files if you want to investigate.
> 
> **The details** I installed the new alpha bios yesterday. Did do the home calibration and tried to print a model with 2 colors, whereas the minor color was in Head [#1](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1) (Text) and the main model itself was in Head [#2](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2). The model itself takes about 1.5 hours to print. Nothing Special as I have printed this over the past 2 days 2 times already.
> 
> 1. Calibrated Home.
> 2. Started to print. Head [Ignore CLion project configuration files. #1](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1) was pre-purging at the sheet.
> 3. Going back home to pick up Head [Fix line endings. #2](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2) produced a head crash as the head was not parked properly (never happened before)!
> 4. I fixed the crash and it was re-homing. Then picked up Head [Fix line endings. #2](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2) and started drawing.
> 5. Then - when head changes happened, I didn't see any new head crashes - all was good.
> 6. However - every layer change to Head [Fix line endings. #2](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2) was producing a noise from the head and then a "filament stuck" message. (I guess that the retracting has problems there?) I couldn't observe this on Head [Ignore CLion project configuration files. #1](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1)  - only on [Fix line endings. #2](https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/2). But it was consisten and every layer needed a 5 times click to restart printing.
> 
> Today I tried it again:
> 
> 1. Tried to pick and park tools via menu. Did do that at least 30 times and there were no problems. All good.
> 2. Re-calibrated Home once more as well as other things.
> 3. Tried again pick and park - still - all good.
> 4. Started to print the same thing again with two other heads to exclude the possibility of the head being the problem. The result: After the pre-purging of the filament another head crash when switching to the head which does the first printing to do the pre-purge. And then - a Blue screen of death. Never have seen that. I have stored the dump on my USB drive and the printer rebooted. Just to get another blue screen of death.
> 
> I am not sure what to do now. Any help is appreiciated!
> 
> ### How to reproduce
> Please see in description.
> 
> Install latest Alpha BIOS and try to print. Does not work for me
> 
> ### Expected behavior
> It prints - just like it always does.
> 
> ### Files
> _No response_

Have same Problem.

Every time a Head Crash, when tool gets changed back to tool one.  Tool 2 crash in the dock. --> Blue Screen.
Just while a print, where 2 heads are used.   If i use the menu to pick and park the toolheads --> no problem.

ulab, 2025-09-30:
Having the same issue with tools getting parked resulting in a crash. I tried a complete calibration run too with no luck.

In addition the Z calibration runs fine, gives me a green check mark and then reboots into a red screen of death "Homing Error Z" (17301)

6.4.0-alpha+11774

3d-gussner, 2025-10-01:
FYI: The developers are working on these issues.