prusa3d/Prusa-Firmware-Buddy #4845: [BUG] Crash into Extruder 1

Hubertusfan, 2025-09-19:
### Printer model

XL 5t

### Firmware version

6.2.6+8934

### Printing from...

PrusaLink

### Describe the bug

A reverse movement is performed behind the Y 0 position. This causes the tool changer to crash with extruder 1.

### How to reproduce

After cancelling a print with extruder 5 due to poor adhesion, I park the extruder by entering the menu. After parking the extruder, the tool changer moves to the left side.
I then restart the print. Now the tool changer rams extruder 1 at full force with a loud noise! After that, normal homing is performed.

### Expected behavior

This must not happen! The machine knows the position of the tool changer. Why does it move back beyond the zero position? Why is a reference run performed at all? Because the printer knows the position of the tool changer, homing is completely unnecessary.

In general, this printer performs a lot of unnecessary homing, even on many other occasions.

murk-sy, 2025-09-20:
Related issue, though I only mentioned the axes being confused in passing: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4812

I've seen weird behaviour with reversed movement direction even before v6.4.0-alpha but it seems to be exacerbated with the alpha

gjpc, 2025-09-20:
Same type of head crash T2 Back2 up into T3 during a head swap on 6.26
Model and crash.bin attached

[dump.zip](https://github.com/user-attachments/files/22440295/dump.zip)

[skeletonmeditating_0.4n_0.2mm_PLA,PLA,PLA_XLIS_2h2m.zip](https://github.com/user-attachments/files/22440310/skeletonmeditating_0.4n_0.2mm_PLA.PLA.PLA_XLIS_2h2m.zip)

gjpc, 2025-09-20:
I have now gone back 3 release versions of XL firmware and every version of the firmware causes the T2 into T3 crash printing the above file.

I thought a little about this and decided to move the wipe tower away from the printed object. The printer stopped crashing heads on 6.2.1. So I reloaded 6.2.6 and the head crash did not occur.

@Hubertusfan : do we need new issue, or will this comment be sufficient to investigate the head crash being caused by wipe tower position? 

gcode that does not cause the head crash:
[skeletonmeditating_0.4n_0.2mm_PLA,PLA,PLA_XLIno head crash S_2h5m.zip](https://github.com/user-attachments/files/22443776/skeletonmeditating_0.4n_0.2mm_PLA.PLA.PLA_XLIno.head.crash.S_2h5m.zip)

Hubertusfan, 2025-09-20:
I don't think my problem is caused by a wipe tower. I didn't print one during my process.
The error occurs for me after cancelling a print and subsequent homing.

closed 2025-12-01 by github-actions[bot]