prusa3d/Prusa-Firmware-Buddy #4847: [BUG] Core One/6.4.0-alpha: Chirping motor noises and layer shifts on small circular motions

MartinEmrich, 2025-09-21:
### Printer model

Core One

### Firmware version

6.4.0-alpha

### Upgrades and modifications

Bogey Idler extruder mod

### Printing from...

PrusaLink (OrcaSlicer)

### Describe the bug

Reproducibly, the flow rate test from OrcaSlicer (2.3.1-beta), while trying to calibrate flow rate, does occasional "chirping" motor noises, including slight layer shifts.
It does not happen on infill, layer crossings or the other usual suspects for actual nozzle collisions.

But it happens most intensely during the smallest circular motion when printing the "archimedian circles". It might be some interaction between that G-Code.
I made a video, I am quite sure this is not a material collision: https://www.youtube.com/watch?v=dY0oNIY09ls
 (noise at 0:14, 0:44, 1:44). Sometimes it is so strong that it causes layer shifts.
Attached are 3mf (IDK if this contains all the "magic" injected for calibration prints) and a G-Code export.

Filament is ASA, 115°C bed, 285°C nozzle, 50°C chamber. (Other, longer prints for "normal" parts from PrusaSlicer were no issue, so I don't think it's a thermal/motor hardware issue either)

I have a feeling this might be where from G-Code motions the motor controlling voltages/timings/currents are calculated, and some formula actually produces these signals leading to the noises.... If this is impossible, feel free to close the issue.

[chirping-core-one-flowrate-test.zip](https://github.com/user-attachments/files/22454604/chirping-core-one-flowrate-test.zip)

### How to reproduce

Core One: 6.4.0-alpha
OrcaSlicer 2.3.1-beta, Core One 0.4 (non-HF) profile, 0.2mm SPEED print profile. Flow rate calibration "YOLO".

MartinEmrich, 2025-09-21:
Second try, at the last layer with the big archimedian spirals, it got bad enough for a layer shift, and I had to abort.
Note the weird motor noise even after aborting the print, here's a second video: https://www.youtube.com/watch?v=NQnBWg_K0ZM
During the last homing motion, it also did not collide with the rear end stop, this happened "mid-air".

3d-gussner, 2025-09-22:
@MartinEmrich Can you please try to print this sliced by PrusaSlicer 2.9.3, especially the archimedian spiral.
Listening to your video it looks like the acceleration is really high.
Did it work with previous stable firmware?

MartinEmrich, 2025-09-24:
It still gets more interesting: I printed the same without Arc Fitting, and validated that there are no G2/G3 commands in the G-Code file. Yet it looks and sounds just as choppy and weird.

Then I exported the 3MF and opened in PrusaSlicer. Of course all settings are gone, and I just switched to Archimedian Chords surface pattern.
PrusaSlicer generates them using *only* G2 instructions, and Core One prints the circles in one smooth circular motion.

# Results so far

| FW     | Slicer| Print  | G-Code Arcs   | Result  |
| ------ | ----- | ---- | ------ | ----------- |
| alpha  | Orca  | Hot  | Arcs   | Yes, Severe Chops https://www.youtube.com/watch?v=dY0oNIY09ls |
| alpha  | Orca  | Cold | Arcs   | Yes, Severe Chops |
| stable | Orca  | Hot  | Arcs   | Yes, but more subtle: https://www.youtube.com/watch?v=NXzqDHpuOdU |
| stable | Orca  | Hot  | NoArcs | Yes, subtle, identical to with Arcs |
| stable | Prusa | Hot  | yes   | No, Circles printed smooth in constant-speed motion  |

* Temperature/Filament is *irrelevant*
* Appears both in alpha and stable firmware, just way more subtle in stable
* Happens with OrcaSlicer G-Code, irrespective of actually using G2/G3.
* No issue with PrusaSlicer-generated G-Code

One thought: Maybe the OrcaSlicer G-Code assembling the spirals from so many tiny linear G1 movements is just too much for the "motion-planning" part in Buddy-Firmware, which makes it first stall, then try to "catch up" with unhealthy acceleration or motor instructions? One tile-surface segment is ~5000 lines of G-Code. Meanwhile PrusaSlicer generates them from only G2, resulting in ~400 lines.

(Indeed for Klipper-based printers, the recommendation is the other way around; They recommend to disable Arc support, as Klipper motion planner breaks it up in tiny linear motions anyways).

While the obvious short-term fix is "Just use PrusaSlicer", I hope my expectation of Core One being capable of G-Code from other major well-regarded slicers, too is reasonable.

Next: back to 6.4.0-RC

3d-gussner, 2025-09-25:
@MartinEmrich Thanks, Dankeschön for the detailed test and analysis.

MartinEmrich, 2026-10-07:
Any chance this has reappeared in a different form in 7.0.0?

After upgrade, I printed a round object (indeed from PrusaSlicer 2.9.6 this time) and heard again very ugly noises on the circular path. not even 2cm height in, I had a major layer shift, and had to abort...
Will downgrade and retry, if that fixes it, I open a new issue.