prusa3d/Prusa-Firmware-Buddy #4862: [BUG] Reloading filament after purge line

peterbenedicic, 2025-09-24:
### Printer model

MK4S MMU3

### Firmware version

6.4.0 alpha, RC

### Upgrades and modifications

None

### Printing from...

USB drive, Prusa connect

### Describe the bug

At the start of the print, when filament goes into nextruder at first there are about 3 skipped steps on MMU3 (before MMU3 disengage; when filament is allready in nextruder), than when filament goes into the nextruder just before start of purge there are 2-3 skipped steps of extruder motor. Purge line is than normally printed and than it ends with "Reloading filament" and yellow led light, that stays at 99% forever. You have to reset the printer to resolve problem.

### How to reproduce

Start normal print with MMU3 unit.

### Files

[image]

jariascr2021, 2025-09-26:
another over here, same issue.

Mikyner, 2025-09-26:
I have encountered this problem with both alpha and RC

peterbenedicic, 2025-09-26:
Yes, RC version did nit solve the problem.

3d-gussner, 2025-09-26:
Thanks for such quite answers. Forwarding that information to the developers.

Pitty it happens _maybe 50% of all prints_ which makes it difficult to see why. It may be the reason why the internal tests passed.

rafibar, 2025-10-01:
I have encountered this problem with both alpha and RC

rafibar, 2025-10-01:
I return to the latest stable firmware. I am waiting to the next stable version. Someone knows when it will be ready?

rafibar, 2025-10-02:
I also switch back to the latest stable firmware. I hope it will not appear again. 
I' m looking forward for a new firmware that correct this specific issue.

sacrelege, 2025-10-18:
Same issue here (~50% of all prints)
Hardware: MK4S + MMU3

Barney15e, 2025-10-23:
I posted the same bug earlier: [https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4834](url)
I only noticed it happening when I sent the job through Prusa Connect. However, if it is 50% problem, that might explain why it printed correctly from the panel after the failure via Prusa Connect.

beroliv, 2025-10-25:
same issue here, but i send my files always direkt to the printer.

rafibar, 2025-10-25:
> Same issue with MK4s and MMU3.
> It happend more often while starting Print via Prusa connect, but happend
> also while staring prints direkt on the printer.

I don’t understand why Prusa is ignoring these complaints and not providing
any response regarding the expected fix for this bug.

adam2104, 2025-11-02:
I reverted to 6.2.6 as well. I was seeing this on 50% of prints as well with no clear cause.

relayer111, 2025-11-05:
MK4s and MMU  same problem

3d-gussner, 2025-11-18:
@guneremre The testing of the new firmware for MINI, MK3.5, MK3.9/S, MK4/S, XL, CoreOne and CoreOne L is complex and takes some time. A fix for one specific printer may (shouldn't but can) cause issues on another. The developers and testing team are working on it.

We have different motion systems, motors, electronics, etc. this takes time.

guneremre, 2025-11-20:
Issue still persists in RC2 firmware.

mjablon1, 2025-11-22:
+1 from me - also encountered this bug in alpha and RC (about 50% of the times, when printing starts).
Keeping fingers crossed for the devs to fix this in final 6.4.x
As for now, downgrading to 6.2.6

beroliv, 2025-12-06:
Issue is gone since 6.4.0. MK4/s MMU3

guneremre, 2025-12-06:
> Issue is gone since 6.4.0. MK4/s MMU3

mine still persist with MK4s with MMU3

LarGriff, 2025-12-11:
Still happening with 6.4.0. after filament swap.  Mk4s/mmu3

[image]

juergenbr, 2025-12-25:
I keep running into the same issue the last few days. Print randomly crashing in single material mode with "99% parking" displayed. Currently testing with downgraded firmware (6.2.6 on the MK4S and 3.0.3 on the MMU3).

peterbenedicic, 2025-12-26:
> I keep running into the same issue the last few days. Print randomly crashing in single material mode with "99% parking" displayed. Currently testing with downgraded firmware (6.2.6 on the MK4S and 3.0.3 on the MMU3).

Hi, try to recalibrate filament sensor on nextruder.

juergenbr, 2025-12-30:
> > I keep running into the same issue the last few days. Print randomly crashing in single material mode with "99% parking" displayed. Currently testing with downgraded firmware (6.2.6 on the MK4S and 3.0.3 on the MMU3).
> 
> Hi, try to recalibrate filament sensor on nextruder.

Unfortunately that did now work. Downgrade fixed the issue and I am able to print without issues again.

Barney15e, 2026-01-08:
When I reinstalled the 6.4.0 release, I also updated the firmware in the MMU3. It appears it is the same version as prior, but I went ahead and installed it, too. I have not had the reloading problem since. I haven't done a lot of printing, but I was seeing the problem regularly on most print jobs. 
I had also thought it might be related to Stuck Filament detection, so I had that off. I was getting some bad first layer adhesion and couldn't figure out why, so I re-enabled Stuck Filament Detection to see if it was stuck filament. I have not had any spurious Stuck Filament warnings, either. The bad first layer may have been my colder than normal room temperature. I haven't seen that since it warmed up a little.

3d-gussner, 2026-01-09:
FYI: This issue should be solved in the next firmware release

3d-gussner, 2026-01-13:
@LukeShootsThings As @Ro3Deee mentioned the internal development isn't public until a release here.

GwynnePrints, 2026-02-09:
Imagine marking as issue as “solved” when it isn’t actually resolved yet, and just saying it will be fixed in the future…

Mikyner, 2026-02-09:
The only FW files I have on the printer's disk are MK4_MK4S_MK3.9_MK3.9S_firmware_6.2.6.bbf
MMU3_FW3.0.3+896.hex

 I don't have 6.4.0 on the flash drive.
So I'm afraid that's not the way to go.

anghelos, 2026-02-10:
> It would be nice if someone from Prusa let us know if and what to expect
> on this issue to manage our expectations.
> If we are only (e.g.) 0,1% of the users that have a problem this specific,
> I don’t expect any action…😉.
> I reply with some insights would help at least from their side.
>
> Best regards!
>
> @Prusa-Support <https://github.com/Prusa-Support>

For what it's worth, I've had this issue with the alpha, but haven't
encountered it on either of my two MK4Ss with MMU3 since the 6.4.0 release.

Barney15e, 2026-02-11:
I’m not sure if I posted this here on this issue, but I had success updating the MK4S firmware then updating the MMU3 firmware. The MMU3 firmware was the same version I already had installed, but since doing both, I don’t see the problem.

Benjamin-Morrison-Hexagon, 2026-02-13:
I am currently on 6.4.0 and experiencing this issue. I contacted support about it and they basically just said "good luck". It's unbelievably frustrating for a software issue to cause failures, not hardware.

mylife4aiurr, 2026-03-10:
Has any1 checked if this bug is fixed in new firmware 6.5.3-beta or new mmu3 firmware?

anghelos, 2026-03-11:
@markguy100 it's available here: https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v6.5.3-beta

There's also a link in the description to download the MMU firmware 3.0.4. You need a micro USB cable to update that.

markguy100, 2026-03-11:
I have run 2 print jobs, 5 hours in total. No sign of the error.