prusa3d/Prusa-Firmware-Buddy #4866: [FEATURE REQUEST] Delayed print start via printer UI

me25542, 2025-09-27:
### Printer model

All, xBuddy-based if not feasible on MINI

### Describe the feature

The delayed start feature in Prusa Connect is very convenient and useful; it would be nice to have an equivalent or similar feature available offline.

One case when such a feature is useful is to let an enclosure heat soak before the start of a print, by preheating the machine and scheduling a start in a few minutes. Another use case is scheduling a print to start in the morning, without having to be at the printer to start the print.

### Expected functionality

After clicking on a print file, the user would be presented with an additional option such as "schedule print", "delay print", or something similar ("Print" could be renamed "Print now"). If the delayed start option was selected, the printer would prompt the user to either:
- Set a time delay before the print start (Start in <`15`> minutes)
- Schedule a time for the print to start (Start at <`09:45 AM`>)

An option to preheat for the loaded material a specified time before the print start could also be present, and would be a convenient addition.

If the wrong filament (or no filament) was loaded, the printer would display the same warning as it would pre-print in the same situation.

If a delayed start were scheduled:
- The printer could display an icon in the screen header indicating such, perhaps a timer icon
- If the user attempted to begin another print job (print or schedule another file), they would be prompted to cancel either the scheduled or the new job
- After the specified amount of time elapsed, or the set time was reached, the printer would begin the print as if started by the user

me25542, 2026-09-28:
This issue should stay open.

suomi35, 2026-10-28:
Seconding the 'stay open' sentiment 😅