prusa3d/Prusa-Firmware-Buddy #4868: [BUG] Bed keeps dropping repeatedly until it crashes into the base and loses alignment

dm8tbr, 2025-09-29:
### Printer model

Core One

### Firmware version

6.4.0-RC

### Upgrades and modifications

none

### Printing from...

USB Drive

### Describe the bug

During filament change the bed keeps dropping a bit during various steps of the process.
Today I had trouble getting the filament to seat properly and had to go through multiple 'retry' steps.
This was enough to crash the bed eventually into the bottom and what I didn't know, lose alignment.

The next print caused the nozzle to _scrape over the bed repeatedly_ during bed levelling and eventually the printer said it couldn't and if it should realign. After that alignment the bed leveling went fine.
My plate now has some surface scratches…

Vaguely similar to #4836. Maybe?

### How to reproduce

Change filament and during feeding in make sure to retry many times and observe how the bed drops a bit each time and eventually crashes.

### Expected behavior

* Do not lower the bed more than once
* Detect a crash and require recalibration of Z-axis before allowing prints

hyiger, 2025-09-30:
I had a very similar thing happen. Was having difficulties getting a very stiff filament lined up into the extruder so I hit "Retry". Each time I hit retry it would lower the bed. Eventually I hit the reset to get out of it.

pilotltd, 2025-09-30:
Similar here - I now have two smooth lines on my satin sheet where it failed to retract when doing levelling moves!!

dm8tbr, 2025-11-18:
Good news! Doesn't repro on the 6.4.0 RC2 that just came out.

3d-gussner, 2025-11-25:
@pilotltd The printer after a power up doesn't "know" the height and starts with `0` only when it homes Z the printer "knows" where the bed actually is.

pilotltd, 2025-11-25:
@3d-gussner - I realise it doesn't know until homed. The answer is stop all the unnecessary downward Z moves and/or allow any axis to be manually moved despite where the printer thinks it is like commercial CNC machines work. They don't know where they are either until homed or operator inputs position, Most also remember last position before power off and not guess 0.